ChipCE/WiFiMan #2: [Q] how can I access to the ESP and change the settings after connected to the router?

klero2, 2018-08-13:
Is not an issue, is just not enough information.

How can I access to the ESP and change the settings after connected to the router? I have been trying to access to the ESP using the assigned IP from the router and also by the ESP default IP 192.168.4.1 but not luck.

Or it is just not possible to change settings after connected to the router?

ChipTechno, 2018-08-13:
Config page cannot be access after connect to router.You can try to call forceApMode() to skip auto connect after startup.this is the sample code.
I will add the ability to change settings after connected to the router as soon as possible.
<pre>
#include <WiFiMan.h>
Config conf;
void setup() 
{
  //create default object
  WiFiMan wman = WiFiMan();
  //press flash(boot) button within 10 seconds to skip auto-connect 
  pinMode(0,INPUT);
  long startW = millis();
  while((millis()-startW)<10*1000)
  {
    if(digitalRead(0)==LOW)
    {
      wman.forceApMode();
      break;
    }
  }
  wman.start();

  if(wman.getConfig(&conf))
  {
  }
}

void loop() {
}
</pre>

ChipTechno, 2018-08-14:
@klero2 
 I updated the library to v1.0.4 .To change the settings after connected to the router , you can use rebootToApMode() or wman.forceApMode().
Please check "reconfig" in examples.

closed 2018-08-14 by ChipTechno

klero2, 2018-08-15:
I updated the library, I will try it when get some spare time. 

Thank you for your support.