ChipCE/WiFiMan #7: Request: Custom Config

te260701, 2018-11-25:
Hi there,
I am glad that I found this library. It is most complete Wifi Manager. It would be great if it can add custom config without modify WebUI and variables.

What I would suggest if I may:
- Wifi Setting section can be fixed, obviously :)
- The rest should be generate from:
```
{
    config1: {
        "group_name": "MQTT Setting",
        "data": [{
            "name": "mqtt_server",
            "value": "",
            "webLabel": "MQTT Server",
            "webPlaceHolder": "MQTT Server Address",
            "inputType": "text"
        }, {
            "name": "mqtt_password",
            "value": "",
            "web_label": "MQTT Password",
            "web_placeHolder": "MQTT Password",
            "input_type": "password"
        },
        ................................]
    },
    config2: ................
}
```

Thank you,
TL

ChipTechno, 2018-11-26:
Hello te260701.
I will try to find a way to add custom config. It may take some time, but I will try to add it as soon as possible.
Sincerely.

te260701, 2018-11-27:
I have been tinkering what data structure. I think this one would be easy enough for user, like me :)
This should be able to loop thru all data and create function getValue(char* key) to get value from the key.
  
```
struct Data {
  char* key;
  char* value; 
};
struct Config {
  char* id;
  Data* row;
};

Data data[0] = {
  {"server0" , "test server0"},
  {"port0"   , "0000"     }
};
Data data[1] = {
  {"server1" , "test server1"},
  {"port1"   , "1111"     }
};
Config config[] = {
  {"setting0", data[0]},
  {"setting1", data[1]},
};
```

te260701, 2018-11-27:
Yes , I looked at ArduinoJson before. When I tried their Assistant, the output code look complicated (for me). It doesn't look like it can loop thru all keys and values automatically.
```const size_t bufferSize = 2*JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(4) + 2*JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(5) + 5*JSON_OBJECT_SIZE(6) + 680;
DynamicJsonBuffer jsonBuffer(bufferSize);

const char* json = "[{\"group\":\"MQTT Setting\",\"data\":[{\"name\":\"mqtt_server\",\"value\":\"\",\"label\":\"MQTT Server\",\"placeHolder\":\"MQTT Server Address\",\"type\":\"text\",\"require\":true},{\"name\":\"mqtt_port\",\"value\":\"\",\"label\":\"MQTT Port\",\"placeHolder\":\"MQTT Port ie 1234\",\"type\":\"number\",\"require\":true},{\"name\":\"mqtt_username\",\"value\":\"\",\"label\":\"MQTT Username\",\"placeHolder\":\"MQTT Username\",\"type\":\"text\",\"require\":true},{\"name\":\"mqtt_password\",\"value\":\"\",\"label\":\"MQTT Password\",\"placeHolder\":\"MQTT Password\",\"type\":\"password\"}]},{\"group\":\"Blynk Setting\",\"data\":[{\"name\":\"blynk_server\",\"value\":\"\",\"label\":\"Blynk Server\",\"placeHolder\":\"blynk-cloud.com\",\"type\":\"text\",\"require\":true},{\"name\":\"blynk_token\",\"value\":\"\",\"label\":\"Blynk Token\",\"placeHolder\":\"Blynk Auth Token\",\"type\":\"text\",\"require\":true}]}]";

JsonArray& root = jsonBuffer.parseArray(json);

const char* root_0_group = root[0]["group"]; // "MQTT Setting"

JsonArray& root_0_data = root[0]["data"];

JsonObject& root_0_data0 = root_0_data[0];
const char* root_0_data0_name = root_0_data0["name"]; // "mqtt_server"
const char* root_0_data0_value = root_0_data0["value"]; // ""
const char* root_0_data0_label = root_0_data0["label"]; // "MQTT Server"
const char* root_0_data0_placeHolder = root_0_data0["placeHolder"]; // "MQTT Server Address"
const char* root_0_data0_type = root_0_data0["type"]; // "text"
bool root_0_data0_require = root_0_data0["require"]; // true

JsonObject& root_0_data1 = root_0_data[1];
const char* root_0_data1_name = root_0_data1["name"]; // "mqtt_port"
const char* root_0_data1_value = root_0_data1["value"]; // ""
const char* root_0_data1_label = root_0_data1["label"]; // "MQTT Port"
const char* root_0_data1_placeHolder = root_0_data1["placeHolder"]; // "MQTT Port ie 1234"
const char* root_0_data1_type = root_0_data1["type"]; // "number"
bool root_0_data1_require = root_0_data1["require"]; // true

JsonObject& root_0_data2 = root_0_data[2];
const char* root_0_data2_name = root_0_data2["name"]; // "mqtt_username"
const char* root_0_data2_value = root_0_data2["value"]; // ""
const char* root_0_data2_label = root_0_data2["label"]; // "MQTT Username"
const char* root_0_data2_placeHolder = root_0_data2["placeHolder"]; // "MQTT Username"
const char* root_0_data2_type = root_0_data2["type"]; // "text"
bool root_0_data2_require = root_0_data2["require"]; // true

JsonObject& root_0_data3 = root_0_data[3];
const char* root_0_data3_name = root_0_data3["name"]; // "mqtt_password"
const char* root_0_data3_value = root_0_data3["value"]; // ""
const char* root_0_data3_label = root_0_data3["label"]; // "MQTT Password"
const char* root_0_data3_placeHolder = root_0_data3["placeHolder"]; // "MQTT Password"
const char* root_0_data3_type = root_0_data3["type"]; // "password"

const char* root_1_group = root[1]["group"]; // "Blynk Setting"

JsonObject& root_1_data0 = root[1]["data"][0];
const char* root_1_data0_name = root_1_data0["name"]; // "blynk_server"
const char* root_1_data0_value = root_1_data0["value"]; // ""
const char* root_1_data0_label = root_1_data0["label"]; // "Blynk Server"
const char* root_1_data0_placeHolder = root_1_data0["placeHolder"]; // "blynk-cloud.com"
const char* root_1_data0_type = root_1_data0["type"]; // "text"
bool root_1_data0_require = root_1_data0["require"]; // true

JsonObject& root_1_data1 = root[1]["data"][1];
const char* root_1_data1_name = root_1_data1["name"]; // "blynk_token"
const char* root_1_data1_value = root_1_data1["value"]; // ""
const char* root_1_data1_label = root_1_data1["label"]; // "Blynk Token"
const char* root_1_data1_placeHolder = root_1_data1["placeHolder"]; // "Blynk Auth Token"
const char* root_1_data1_type = root_1_data1["type"]; // "text"
bool root_1_data1_require = root_1_data1["require"]; // true```

Serialize even longer code lol.

ChipTechno, 2018-11-27:
i'm not sure if it can loop thru all the keys or not , but try this
<code>
String keyValue = jsonObj["keyName"].as<String>();</code>

ChipTechno, 2018-11-27:
'
String keyValue = jsonObj["keyName"].as&lt;String&gt;();'

ChipTechno, 2018-12-04:
Hello. 
I added "custom config" function. You can download the newest version form <a href="https://github.com/ChipTechno/WiFiMan/tree/testing">testing branch</a>.
please check the example "customConfig" for usage.
Thank you!

closed 2018-12-04 by ChipTechno

te260701, 2018-12-05:
I got error:

~\Arduino\libraries\WiFiMan\src\DebugHelper.cpp:3:1: error: 'DebugHelper' does not name a type

 DebugHelper::DebugHelper()

 ^

~\Arduino\libraries\WiFiMan\src\DebugHelper.cpp:10:1: error: 'DebugHelper' does not name a type

 DebugHelper::DebugHelper(bool enable)

 ^

~\Arduino\libraries\WiFiMan\src\DebugHelper.cpp:24:6: error: 'DebugHelper' has not been declared

 void DebugHelper::println(String msg)

      ^

~\Arduino\libraries\WiFiMan\src\DebugHelper.cpp:24:27: error: variable or field 'println' declared void

 void DebugHelper::println(String msg)

                           ^

~\Arduino\libraries\WiFiMan\src\DebugHelper.cpp:24:27: error: 'String' was not declared in this scope

ChipTechno, 2018-12-05:
Hello.
Debughelper has been removed in the newest version.Please remove all files in <home folder>/Arduino/libraries/WiFiMan/ and replace with files from my testing branch.

ChipTechno, 2018-12-05:
Thank for advice!
 I will add input label and no-mqtt flag as soon as possible.   
 Load existing value is possible.But the library is uses Arduino's String object.The more changes maked to String object, the more waste memory and heap fragment become.So i may unable to add that function to library.

ChipTechno, 2018-12-06:
Hi te260701.
I upload a new version in testing branch.
I added input label and addition fields as your request. The load existing value will take some time, I need to test the memory consumption before decide if i should add it or not.
Btw, blynk look very nice. I will try it in my next project.