ChipCE/WiFiMan #8: Is this the new name for esp8266-ota?

njh, 2018-11-27:
It noticed that this repo has vanished:
https://github.com/chiptechno/esp8266ota

Is WifiMan the new name for esp8266-ota?

ChipTechno, 2018-11-27:
Not quite, esp8266ota is now a component of wifiman.You can include <code>#include &lt;ESP8266OTA.h&gt;</code> and use the library as normal.
I will re-add example for esp8266 ota as soon as possible.
Sorry for inconvenience!

njh, 2018-11-28:
Ok, thanks.

It might have been good to have kept the esp8266ota repository, but deleted everything apart from the README, explaining why it has disappeared.

closed 2018-11-28 by njh

ChipTechno, 2018-11-28:
The OTA and wifiman library use the same header file for webUI .When rework on memory allocation, i notice that duplice make the sketch use more memory than usual.So i merge it wifiman.I will add documents for ESP8266OTA when im back from work.
Regard.

ChipTechno, 2018-11-28:
Hello.
I just push the newest version 1.1.2. It will available in library manager after an hour.
Sorry for the inconvenience!