ChipCE/WiFiMan #11: reconnect after wifi down

te260701, 2019-01-23:
Hi Chip,

I have an issue when wifi was down and esp8266 reboot into AP config mode and it seem never get out of it even after portal timeout (maybe I'm wrong). I have to manually press reset button to get it working again.
It is weird. It lost connection at 11:58pm everyday and sometime can reconnect but most of the time it cannot. I check my router actually has not set to reboot everyday hmm...

Is there a way to reboot normally (autoconnect) after config portal timeout?

Thank you,
Sam

ChipTechno, 2019-01-24:
By default, the device will auto connect to WiFi with saved config.
What version of esp8266 are you using? This library heavily depends on flash memory space, it may not save config on some low flash esp8266(esp01 for example).
Can you provide the debug log, please?

te260701, 2019-01-25:
Hi Chip,

It was my bad. There is bug in my code that call for a function that doesnt exist.

Thank you for your very fast reply,

closed 2019-01-25 by te260701