ChipCE/WiFiMan #12: Permantenly access to configuration web page and http ota upgrades ?

jirmjos, 2019-04-24:
Hello chip
First of all, thanks for sharing your library for Wi-Fi administrator connections and I think this is the best one I know right now (I tried a really huge group of them for ESP8266 and ESP32 with such a different success rate)) and for me it is the best library. They have covered the most important features with a really functional reliability.
I did tests with the provided examples and, until now, I see that it is very reliable when I am working and that it is really easy to manage and customize.

The only thing that I have not managed to configure properly is that it can remain accessible from web configuration (maybe the main index.html page?) To reconfigure parameters (which are active at that time) and to be able to perform a (online) firmware upgrade by http-OTA but all this when it is working in "normal mode" connected to wifi station ap.
Surely it is my stumble but none of the provided examples I can get it configured that way.
Any hint will be awesome.

Thank you

ChipTechno, 2019-06-03:
Hello jirmjos,
Sorry for the late response. 
 The reason for the web interface cannot be accessed in STA mode is the web interface use almost of the device memory, so the ideal is after the web interface complete its job it will be destroy and return all the free memory for the main program. 
 If you need to remote set the device into config-mode during STA mode you can call rebootToApMode() in the loop() function. I will add an example for this function soon.
Sincerely.

ChipTechno, 2019-06-03:
Hi jirmjos,
I added example for remote reset ( reboot to config mode) via mqtt messenger. Hope that will help.
Sincerely.

jirmjos, 2019-06-03:
Hi Chip,

Thank you so much for your reply.

I agree about web interface is very ungry on resources, so your approach will be the right path, but as per my tests mostly of boards i´m using have more than enough memory (4M) to mantain propper resources to main tasks while still web interface running if web files are stored on spiffs. 
From the start I´m using the rebootToApMode(), to reconfigure, but with over all the wole process the device are offline and that is not the best mainly if only need change the mqqt password or broker server or if need refresh firmware over ota (that only need a final reboot when it´s finish)
On my daily use bases I concluded that a permanent access on STA normal working mode to web configuration interface (OTA and MQTT parameters) and maybe is not a must but for me is , so maybe at least have the option to turn the web interface to on or off in STA mode would be great evaluate and I think can be do it with not so much changes in the library.
Again, thank you and congratulations for this good work.

Jose

ChipTechno, 2019-06-03:
Thank for reply!

I will try to add the function to keep the web interface alive in the next versions.    
Define WiFiMan in global scope , comment line 407 and 407 in WiFiMan.cpp , and find a way to call webServer->handleClient(); in loop function may solve the problem. But i need to test it before release the new version.   
By the way 4M is the flash size where the program and config files are stored. The random access memory of esp8266 is only about 40KB .
Sincerely.