ChipCE/WiFiMan #13: its too slow if to connect if you use LCD I2C library

flexifi, 2019-06-08:
#include <WiFiMan.h>
#include <LiquidCrystal_I2C.h>

Config conf;
LiquidCrystal_I2C lcd(0x27, 16, 2);

void setup()
{
  Serial.begin(115200);
  lcd.init();
  lcd.backlight();
  lcd.begin(16, 2);
  //create default object
  WiFiMan wman = WiFiMan();
  wman.start();

  if (wman.getConfig(&conf))
  {
    //display device status
    Serial.println("Connected to AP");
    Serial.print("SSID : ");
    Serial.println(conf.wifiSsid);
    Serial.print("Passwd : ");
    Serial.println(conf.wifiPasswd);
    Serial.print("MQTT server : ");
    Serial.println(conf.mqttAddr);
    Serial.print("MQTT password : ");
    Serial.println(conf.mqttPasswd);
    Serial.print("MQTT username : ");
    Serial.println(conf.mqttUsername);
    Serial.print("MQTT Id : ");
    Serial.println(conf.mqttId);
    Serial.print("Sub topic : ");
    Serial.println(conf.mqttSub);
    Serial.print("Pub topic : ");
    Serial.println(conf.mqttPub);
    Serial.print("MQTT port : ");
    Serial.println(conf.mqttPort);
    Serial.print("Master password : ");
    Serial.println(conf.masterPasswd);
    Serial.print("Device mDNS name : ");
    Serial.println(conf.mdnsName);
    Serial.print("IP : ");
    Serial.println(conf.localIP);
  }
}

void loop() {
  //
}

when i use this code it took 16 secs before it connects

ChipTechno, 2019-06-18:
The esp8266 is a single thread MCU, it can only do one thing at a time. After wman.start() is called the esp will go into a loop to handle the portal server until it can connect to AP or timeout. It cannot handle the portal and the LCD at the same time unless you modify the code and add LCD handler into the portal loop. It is better to init the LCD code (lcd.init(); lcd.backlight(); lcd.begin(16, 2);) after wman.start(). 
 There is a small delay between each connect attempt, the esp rarely connects at the first try. How long it takes when you connect without LCD code? Does it connect faster when using other libraries/other methods?

ChipTechno, 2019-07-11:
setConnectDelay(ms) function has been added in v1.4.0. Hope this will help to solve your problem.

closed 2019-07-11 by ChipTechno