ChipCE/WiFiMan #15: Continuous watchdog resets during wifiman.start() in release 2.4.1

DerTomm, 2019-11-16:
Hi ChipTechno,

I just updated Wifiman in my application to version 2.4.1 and realized that there seems to be an issue in this release as I am constantly getting "Soft WDT" resets.
To avoid any data incompatibilities I erased the entire flash and tested with your "Simple" example (simple.ino).

This is the error message:

```
Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffa70 end: 3fffffc0 offset: 01b0
3ffffc20:  012bf8c9 3ffffcf0 00000006 00000003
3ffffc30:  4024158f 00000000 3ffeeb58 3ffffc80
3ffffc40:  3ffeeb58 3ffffcf0 000104ee 40203f09
3ffffc50:  3ffeeb58 00000006 3ffffcf0 40203fcf
3ffffc60:  3fff019c 3ffffc98 3ffffcf0 00002463
3ffffc70:  3ffeeb58 3ffffc98 3ffffcf0 40206f97
3ffffc80:  3fff04ac 0017001f ff006469 616d754b
3ffffc90:  0074656e 00fef998 3fff034c 0010001f
3ffffca0:  fffffc34 00000000 3ffffbe4 3ffffc04
3ffffcb0:  3ffffd90 3fff019c fffeeb90 4020752a
3ffffcc0:  00000000 00000000 3ffffcf0 3ffeecb0
3ffffcd0:  3fffdad0 00000000 3ffffcf0 402070b2
3ffffce0:  3fffdad0 00000000 3ffeeb90 4020122a
3ffffcf0:  00010000 ffffffff feefef00 ffffffff
3ffffd00:  feefef01 00000000 feefef00 000003e8
3ffffd10:  feefeffe 000001f4 00000001 00000001
3ffffd20:  feefef00 00000024 0000000f feefef35
3ffffd30:  38707365 2d363632 00006469 00000000
3ffffd40:  00000000 00efeffe 00000000 00000000
3ffffd50:  00efeffe 73736170 64726f77 00efef00
3ffffd60:  696d6461 0000006e 00efeffe 402161e8
3ffffd70:  0101a8c0 402161e8 0101a8c0 402161e8
3ffffd80:  00ffffff 38707365 00363632 00efeffe
3ffffd90:  616d754b 0074656e 00efeffe 3fff04d4
3ffffda0:  0017001f ffefeffe 3fff04fc 000e000f
3ffffdb0:  ffefeffe 33383831 00000000 00efeffe
3ffffdc0:  00000000 00000000 00efeffe 00000000
3ffffdd0:  00000000 00efeffe 00000000 00000000
3ffffde0:  00efeffe 00000000 00000000 00efeffe
3ffffdf0:  3fff0dfc 000e000f ffefeffe 00000000
3ffffe00:  00000000 00efeffe 38505345 00363632
3ffffe10:  00efeffe 3ffefb8c 000d000f ffefeffe
3ffffe20:  6c697542 203a2064 00efef00 6e617242
3ffffe30:  3a206863 00ef0020 3ffefbbc 000f000f
3ffffe40:  ffefeffe 3ffefba4 000f000f ffefeffe
3ffffe50:  3ffefbd4 002b002f ffefeffe 3fff019c
3ffffe60:  3ffefc0c 3ffefcfc 00000000 00000000
3ffffe70:  00efeffe 00000000 00000000 00000000
3ffffe80:  00efeffe 00000000 00000000 00efeffe
3ffffe90:  00000000 00000000 00efeffe 00000000
3ffffea0:  00000000 00efeffe 00000000 00000000
3ffffeb0:  00efeffe 00000000 00000000 00efeffe
3ffffec0:  00000000 00000000 00efeffe 00000000
3ffffed0:  00000000 00efeffe 00000000 00000000
3ffffee0:  00efeffe 00000000 00000000 00efeffe
3ffffef0:  00000000 00000000 00efeffe 00000000
3fffff00:  00000000 00efeffe 00000000 00000000
3fffff10:  00efeffe 00000000 00000000 00efeffe
3fffff20:  00000000 00000000 00efeffe 00000000
3fffff30:  00000000 00efeffe 00000000 00000000
3fffff40:  00efeffe 00000000 00000000 00efeffe
3fffff50:  00000000 00000000 00efeffe 00000000
3fffff60:  00000000 00efeffe 00000000 00000000
3fffff70:  00efeffe 00000000 00000000 00efeffe
3fffff80:  00000000 00000000 00efeffe 00000000
3fffff90:  00000000 00efeffe feefeffe feefeffe
3fffffa0:  feefeffe 00000000 3ffeec80 4020a3f8
3fffffb0:  feefeffe feefeffe 3ffe8550 40100591
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

and so on and so on...

After installing 2.4.0 again this problem does not occur again.

The environment is an ESP8266 on a default ESP-12F breakout board. The application is developed and deployed using PlatformIO.

Regards,
Thomas

ChipTechno, 2019-11-17:
@DerTomm Thanks for the report.
I will recheck the code and fix it as soon as possible.

ChipTechno, 2019-11-17:
@DerTomm I tried the v1.4.1 on my nodemcu but cannot get the same behavior as yours.
Can you give me the debug log, please? With "debug port enabled" the esp will print extra debug data via the serial terminal.

DerTomm, 2019-11-17:
Here you are :-)
This is the serial debug output with version 1.4.1 and the watchdog resets:
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
#>> start
#>> readConfig
#__ Reading config.json
#__ Json : {"wifiSsid":"<removed>","wifiPasswd":"<removed>","mqttAddr":"<removed>","mqttPort":"1883","mqttUsername":"","mqttPasswd":"","mqttSub":"","mqttPub":"","mqttId":"<removed>","masterPasswd":"<removed>"}
#<< readConfig-end
#__ Trying to connect to AP
#>> clientMode
#>> validConfig
#__ Config OK
#<< validConfig-end
#__ Connectiong to AP using saved config...
#>> connect
#__ Connecting to AP : SSID : <removed>   Password : <removed>
#__ .

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffa50 end: 3fffffc0 offset: 01b0
3ffffc00:  000ea6a5 00000000 3ffefb70 00000001
3ffffc10:  3ffe8d54 0000004b 3ffffbf0 3ffffc80
3ffffc20:  00000000 3ffef888 000015af 40205a61
3ffffc30:  00000000 3ffefb70 3ffef888 40205bb1
3ffffc40:  3ffffc00 00000000 3ffffc80 4020c5b6
3ffffc50:  00000001 3fff08f4 3ffffc80 4020c77c
3ffffc60:  3ffffca0 3ffefb70 3ffef888 3ffefc90
3ffffc70:  00000001 3ffefb70 3ffef888 40205c6b
3ffffc80:  3fff087c 0017001f fffefd08 616d754b
3ffffc90:  0074656e 00000000 3ffffc70 3ffffca0
3ffffca0:  0001c200 3ffefb70 3ffef888 3ffefc90
3ffffcb0:  3ffe965c 3ffefb70 3ffef888 402096e6
3ffffcc0:  fffe0020 3ffffee0 3ffef500 3ffefc90
3ffffcd0:  3ffefb50 3ffffee0 3ffef888 40201981
3ffffce0:  00010000 ffffffff ff003300 ffffffff
3ffffcf0:  feefef01 00000000 feefef00 000003e8
3ffffd00:  feefeffe 000001f4 00000000 00000000
3ffffd10:  feefef00 00000024 0000000f feefef35
3ffffd20:  00000000 00000000 ff006469 00000000
3ffffd30:  00000000 ffefeffe 00000000 00000000
3ffffd40:  ffefeffe 00000000 00000000 ffefef00
3ffffd50:  00000000 00000000 ffefeffe 40219cd8
3ffffd60:  0101a8c0 40219cd8 0101a8c0 40219cd8
3ffffd70:  00ffffff 00000000 00000000 ffefeffe
3ffffd80:  00000000 00000000 ffefeffe 00000000
3ffffd90:  00000000 ffefeffe 00000000 00000000
3ffffda0:  ffefeffe 00000000 00000000 ffefeffe
3ffffdb0:  00000000 00000000 ffefeffe 00000000
3ffffdc0:  00000000 ffefeffe 00000000 00000000
3ffffdd0:  ffefeffe 00000000 00000000 ffefeffe
3ffffde0:  00000000 00000000 ffefeffe 00000000
3ffffdf0:  00000000 ffefeffe 00000000 00000000
3ffffe00:  ffefeffe 00000000 00000000 ffefeffe
3ffffe10:  00000000 00000000 ffefef00 00000000
3ffffe20:  00000000 ffef0020 00000000 00000000
3ffffe30:  ffefeffe 00000000 00000000 ffefeffe
3ffffe40:  00000000 00000000 ffefeffe 00000000
3ffffe50:  00000000 00000000 00000000 00000000
3ffffe60:  ffefeffe 00000000 00000000 00000000
3ffffe70:  ffefeffe 00000000 00000000 ffefeffe
3ffffe80:  00000000 00000000 ffefeffe 00000000
3ffffe90:  00000000 ffefeffe 00000000 00000000
3ffffea0:  ffefeffe 00000000 00000000 ffefeffe
3ffffeb0:  00000000 00000000 ffefeffe 00000000
3ffffec0:  00000000 ffefeffe 00000000 00000000
3ffffed0:  ffefeffe 00000000 00000000 ffefeffe
3ffffee0:  00000000 00000000 ffefeffe 00000000
3ffffef0:  00000000 ffefeffe 00000000 00000000
3fffff00:  ffefeffe 00000000 00000000 ffefeffe
3fffff10:  00000000 00000000 ffefeffe 00000000
3fffff20:  00000000 ffefeffe 00000000 00000000
3fffff30:  ffefeffe 00000000 00000000 ffefeffe
3fffff40:  00000000 00000000 ffefeffe 00000000
3fffff50:  00000000 ffefeffe 00000000 00000000
3fffff60:  ffefeffe 00000000 00000000 ffefeffe
3fffff70:  00000000 00000000 ffefeffe 00000000
3fffff80:  00000000 ffefeffe 00000000 00000000
3fffff90:  ffefeffe 2d6e7474 65646f6e 0000332d
3fffffa0:  3fffdad0 00000000 3ffefc60 4020d8b4
3fffffb0:  feefeffe feefeffe 3ffe8554 40100591
<<<stack<<<
```
In comparison this is the output with version 1.4.0:
```
SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
#>> start
#>> readConfig
#__ Reading config.json
#__ Json : {"wifiSsid":"<removed>","wifiPasswd":"<removed>","mqttAddr":"<removed>","mqttPort":"1883","mqttUsername":"","mqttPasswd":"","mqttSub":"","mqttPub":"","mqttId":"<removed>","masterPasswd":"<removed>"}
#<< readConfig-end
#__ Trying to connect to AP
#>> clientMode
#>> validConfig
#__ Config OK
#<< validConfig-end
#__ Connectiong to AP using saved config...
#>> connect
#__ Connecting to AP : SSID : <removed>   Password : <removed>
#__ .
#__ Connected to AP
#__ IP address : 192.168.65.133
#<< connect-end
#__ Connected to A
P#<< clientMode-end
#__ Connected to AP
#<< start-end
´´´

ChipTechno, 2019-11-18:
I revert some change in v1.4.1, can you check if the WDT still occurred in this branch please.
[issues#15-debug-branch](https://github.com/ChipTechno/WiFiMan/tree/issues%2315)

DerTomm, 2019-11-21:
I am planning this for the upcoming weekend - I'll report back.

ChipTechno, 2020-01-14:
I push the new release(v1.4.2).
If you still have problem with the WDT reset, fell free to open the Issue again.
Thank for your report!

closed 2020-01-14 by ChipTechno

DerTomm, 2020-01-15:
Damn - I totally lost this subject from my list. Sorry for this.

Since my son was born the priority and time of my projects have dropped significantly - but I will test the solution as soon as possible.

Thanks for the release and regards,
Thomas

DerTomm, 2020-09-06:
Today I finally wanted to test the new version - but then I noticed that you had to disable the OTA update function due to incompatibilities. Unfortunately OTA updates are an essential function for me and I can't do without them.
Do you think you can make your library compatible again in the near future? I would hate to look for another library - yours is the best compact lib with exactly the necessary features so far.

ChipCE, 2020-09-06:
@DerTomm I cannot promise that I can add it back, but I will give it a look in the next few day.

DerTomm, 2020-12-09:
Carefully asked: Did you already have the chance to look into the OTA update function ("Temporarily disable in v1.4.2 due incompatible with the new ESP8266 core library")? Bringing this feature back in a new release would be great...

ChipCE, 2020-12-10:
@DerTomm I have some ideal to rework the whole lib and move all heavy job to the client side(browser), this will make the UI more friendly and using less space. But I still not have time to code the new version.

DerTomm, 2021-01-02:
As far as I see the necessary change to make WiFiMan compatible with current ESP8266 Arduino releases again is tiny. So I created a PR to bring the feature back in a release v1.4.3.
It would be nice if you could take a look at it and if it works for you too publish v1.4.3 soon.

ChipCE, 2021-01-02:
Thank you! I will test the code tomorrow.