ChipCE/WiFiMan #16: Complie Error

te260701, 2019-12-18:
I can't compile any example. I'm using v1.4.1. 

`Arduino: 1.8.10 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\xxxx\Documents\Arduino\libraries\WiFiMan\src/WiFiMan.h:10:0,

                 from C:\Users\xxxx\Documents\Arduino\libraries\WiFiMan\examples\advance\advance.ino:1:

C:\Users\xxxx\Documents\Arduino\libraries\WiFiMan\src/ESP8266OTA.h:22:7: error: using typedef-name 'using ESP8266WebServer = class esp8266webserver::ESP8266WebServerTemplate<WiFiServer>' after 'class'

 class ESP8266WebServer;

       ^

In file included from C:\Users\xxxx\Documents\Arduino\libraries\WiFiMan\src/WiFiMan.h:5:0,

                 from C:\Users\xxxx\Documents\Arduino\libraries\WiFiMan\examples\advance\advance.ino:1:

C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WebServer\src/ESP8266WebServer.h:241:80: note: 'using ESP8266WebServer = class esp8266webserver::ESP8266WebServerTemplate<WiFiServer>' has a previous declaration here

 using ESP8266WebServer = esp8266webserver::ESP8266WebServerTemplate<WiFiServer>;

                                                                                ^

Multiple libraries were found for "DNSServer.h"
 Used: C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\DNSServer
Multiple libraries were found for "WiFiMan.h"
 Used: C:\Users\xxxx\Documents\Arduino\libraries\WiFiMan
Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi
Multiple libraries were found for "ESP8266WebServer.h"
 Used: C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WebServer
 Not used: C:\Users\xxxx\Documents\Arduino\libraries\WebServer
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\xxxx\Documents\Arduino\libraries\ArduinoJson
Multiple libraries were found for "ESP8266mDNS.h"
 Used: C:\Users\xxxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266mDNS
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

te260701, 2019-12-19:
Esp8266 Board v2.6.3 breaks it. I downgraded board 2.4.2 and ArduinoJson to v6.
It is working now.

I have not try any thing in between 2.4.2 and 2.6.3.
so, @ChipTechno could you please list version number in README?.

ChipTechno, 2019-12-20:
@te260701 Thank you for the report.
Confirmed that the library cannot compile with the new version of Esp8266 core due change of ESP8266WebServer class. I will add the system requirement into README file.
I will try to make the library to works with the new version of esp8266 later.

ChipTechno, 2020-01-14:
I pushed new release v1.4.2. I temporarily disable the OTA update function due it not compatible with new ESP8266 core(v2.6.x). I will try too re-add the function when i have free time.
Thank for the report!

closed 2020-01-14 by ChipTechno